arduino-libraries/RTCZero #4: Incorrect hour rollovers on Adafruit Feather M0 board.

cavemoa, 2015-12-26:
Hi, have just started using your library on the adafruit Feather M0 board which has a SAMD21. 

I get some strange behavior with the roll over of hours when I set the using the simplertc example:

12:00:00 rolls over to 28:00:00 !
28:59:59 then rolls over directly to 17:00:00
It then runs correctly for a few hours to 23:59:59 which rolls over to 24:00:00 and does not increment the day/date.

Another board user has confirmed they have the same behavior.

I'm guessing it must be chipsetup as I presume the issue doesn't arise on the Arduino Zero. Any thoughts?
Cheers Jon

GabrielNotman, 2015-12-27:
The register bit RTC.MODE2.CTRL.CLKREP is being set incorrectly.
That bit should only be set for 12hr mode.

In 12 hour mode, the highest bit is used to specify AM/PM. If this is read as a 24 hour value, it results in all PM values having 16 added to them.

sandeepmistry, 2015-12-31:
@GabrielNotman is this resolved now that #3 has been merged?

(In particular this line https://github.com/arduino-libraries/RTCZero/pull/3/files#diff-6ed0013c42219b7dc3d3d5d163434a8fR55)

GabrielNotman, 2015-12-31:
I haven't test the post merge version.
But, yes It should be.

sandeepmistry, 2015-12-31:
@GabrielNotman yes, looks good to me. I tried the following initial time values with the ```SimpleRTC``` examples:

```arduino
/* Change these values to set the current initial time */
const byte seconds = 55;
const byte minutes = 59;
const byte hours = 11;
```

```arduino
/* Change these values to set the current initial time */
const byte seconds = 55;
const byte minutes = 59;
const byte hours = 23;
```

With ```1.3.1``` saw the same issues @cavemoa noted, with ```1.4.0``` rollover works as intended.

@cavemoa could you please try the latest release (```1.4.0```)?

cavemoa, 2015-12-31:
Yes all seems good post merge to me.

sandeepmistry, 2015-12-31:
@cavemoa great! Thanks for verifying!

closed 2015-12-31 by sandeepmistry