arduino-libraries/RTCZero #6: Multiple alarms

xseignard, 2016-01-09:
Hello,
I would like to know if it is possible to set multiples alarms, and how?

Best regards,

Xavier

GabrielNotman, 2016-01-12:
In MODE 2 (Calendar) the samd21 only supports a single alarm.

xseignard, 2016-01-12:
Hello @GabrielNotman 
So is there any way to handle more than one alarm? e.g. using another mode?
Is there any docs out there?

Regards

Xavier

cavemoa, 2016-01-13:
Hi xseignard,

The SAMD21 RTC does have two other modes but neither is directly supported in the Arduino RTCzero library as far a I can see. They are both counter modes one 16bit and one 32. They are more flexible than the calendar in that they can have multiple compare values (ie multiple alarms). 16bit can have 6 compare values for example

Have a look at http://asf.atmel.com/docs/latest/samr21/html/group__asfdoc__sam0__rtc__count__group.html

It might be worth asking a question on the arduino zero forum to see if anyone has implemented them? Though the example code given doesn't look terrible to translate to Arduino.

Cheers CaveMoa

GabrielNotman, 2016-01-14:
Looking at the SAMD21 datasheet:
http://www.atmel.com/images/atmel-42181-sam-d21_datasheet.pdf

Mode0 (32bit counter) supports a single compare.
Mode1 (16bit counter) supports two compares.
Mode2 (Calendar) supports a single alarm.

This library is designed to provide clock functionality and can only operate correctly when using Mode2. If you need to schedule multiple tasks at varying intervals, I would suggest setting up a regular alarm and feeding that into a timer class which then determines which tasks are due and calls them as needed.

agdl, 2016-01-15:
Yes like suggested by @GabrielNotman the easiest way to have multiple alarms is to have something that can be for example a matrix or better a structure in which you write the alarms you want to have and every time the highest alarm is reached the next one will be setted. I can easily implement it and provide an example. I would prefer to have a fixed maximum number of alarms (let's say 6 instead of having a dynamic one)

agdl, 2016-01-15:
Edit i will do it dynamically with a function called something like setNumberOfAlarms(int numberOfAlarms)

xseignard, 2016-01-15:
Hello everybody!
Thanks for taking the time to provide support.

I need to create a counter, decrementing every minutes, and to provide the ability for the end user to set an alarm (be it one month after the counter started). All of that consuming the lowest power possible to reduce the size of the battery.

I think something like an every minute interrupt will do the trick, with a volatile keeping track of remaining minutes before the end of the counter (e.g. my final alarm).

Thanks for your help!

Best regards

Xavier

agdl, 2016-01-15:
I've added the multiple alarms functionality. Please let me know how it seems to you. If it looks good I will make the new release of the library!

GabrielNotman, 2016-01-15:
I really don't think this should to be part of the core library.
This can be done outside of the library just as easily.
Also how does this work with recurring alarms (the match mask)?

drewfish, 2016-01-15:
I also think that it shouldn't be part of this library, and instead a separate library. A general "alarm multiplexer" library might be useful -- something that would enable multiple software alarms based on a single hardware alarm. (For the Arduino Zero it could be used for the TC or TCC alarms for example.)

agdl, 2016-01-16:
@GabrielNotman @drewfish sorry but I'm not understanding if your comments are referred to my commit or to the request of @xseignard !

drewfish, 2016-01-16:
@agdl I was referring to commit 863adf2. I was thinking it might be nice to have a separate generic library (separate from RTCZero) which is able to use a single hardware alarm and provide multiple virtual alarms based on that. Similar to commit 863adf2, but made generic so that it can be used for any hardware alarm.

However now that I've thought about it more the "can be used for any hardware alarm" part is kinda hard because different alarms have different data types. For example the samd TC alarms can use 8bit, 16bit, or 32bit comparators -- how to represent that in a generic C++ API? It might be difficult to do that in a meaningful way.

GabrielNotman, 2016-01-16:
@agdl I was also referring to commit https://github.com/arduino-libraries/RTCZero/commit/863adf2ce8e02a2f25e40f735db53efd9aaa538e.

It is my opinion that this feature does not belong in the clock library. It can be done externally by just scheduling the next alarm every time an alarm callback is received. 

Additionally, all alarms (except MATCH_YYMMDDHHMMSS) are periodic, with the period varying between one minute to one year. I don't think this new feature will work well with periodic alarms, especially with mixed periods.

I think a better solution for @xseignard is to setup a periodic alarm which then updates a scheduler class. One possible class is RTCTimer: https://github.com/SodaqMoja/RTCTimer

For example, in a recent project I have setup a every minute period alarm to wake the board. I then update the scheduler class which calls any due tasks, and then returns to sleep mode. If nothing is due, it is only awake very briefly. You can have longer periods of sleep, but this will affect the scheduling resolution.

agdl, 2016-01-18:
@cmaglie @facchinm what do you think about this? Have I to revert the commit and insert this feature somewhere else?

agdl, 2016-01-22:
After an internal discussion I reverted this repository to remove the SW multiple alarms implementation. However I have this implementation on my fork https://github.com/agdl/RTCZero/tree/MultipleSWAlarms if someone needs it.

From the discussion on this issue I can understand that this issue can be closed. Please reopen it if needed.

closed 2016-01-22 by agdl

prahu, 2017-02-27:
how can we set multiple alarms(minimum 30) by using RTC3231 and arduino uno.

agdl, 2017-02-27:
@prahu this library uses the internal RTC of SAMD21 based arduino boards. Google if exist an arduino library for that kind of RTC

prahu, 2017-02-27:
you want to say that we can not set multiple alarms by using arduino uno and rtc3231

agdl, 2017-02-27:
I'm saying I don't know the rtc3231 and that this library is not intended for it

prahu, 2017-02-27:
ok

GabrielNotman, 2017-02-27:
@prahu for multiple alarms I would suggest using an alarm which fires at a regular interval (every second, every minute etc). Combine this with a scheduling class and you handle a number of scheduled events.

Here is an example which is designed to run on the Mbili (ATmega 1284p with an onboard DS3231 rtc). It should be compatible with the UNO and an external DS3231 rtc, you will have to update the pins used etc. https://github.com/GabrielNotman/Testing/tree/master/Sleep_Interrupt

prahu, 2017-02-27:
sir in my project i want to set multiple alarms by using buttons(3 buttons, one is for set alarm, second for increment and third to switch on next value).first we set alarms and when it will match to the current time then buzzer turns high.

GabrielNotman, 2017-02-27:
@prahu there is quite a bit more required for such a project. I would suggest perhaps enquiring about this on the Arduino.cc forums. I believe that is a more suitable venue for this discussion.

prahu, 2017-02-28:
ok thanks sir