arduino-libraries/RTCZero #7: Option to

bborncr, 2016-03-19:
By commenting out line 49 (//RTCReset();) in RTCZero.cpp the date/time remains persistent even through firmware updates and manual resets as long as power to the chip is maintained.

So far, it seems to be working just fine.  Is there a potential issue that I should be aware of?

Thanks,

GabrielNotman, 2016-03-24:
I've created a pull request (#8) which among other additions includes an attempt to preserve the existing clock time. It does it slightly different than your suggestion, however, it should have the same result.

agdl, 2016-03-31:
@GabrielNotman thank you for your PR. I'm going to tag a new release to include our changes. @bborncr try it and if it is ok for you close the issue please!

bborncr, 2016-03-31:
We tested the changes and they work fine. You can close the issue.

Thanks!

closed 2016-04-01 by agdl