arduino-libraries/RTCZero #9: standbyMode() needs clocks to be turned off?

madvoid, 2016-05-13:
Hi,

I was reading the SAM D21 datasheet and I noticed that when placed into Standby mode, a significant amount of clocks and peripherals must be disabled, so that the voltage regulator is not overloaded (see section 16.6.2.8). However, the `standbyMode()` function in RTCZero doesn't appear to do that. It sets the deep sleep mask and goes directly to sleep. Am I missing something? Should code be added there to turn peripherals/clocks off? I apologize if this is the wrong place to post this, I can move elsewhere if necessary.

Thanks!

sandeepmistry, 2016-05-31:
Hi @madvoid,

> Am I missing something? Should code be added there to turn peripherals/clocks off?

I think the current approach is fine, because none of the peripherals a enabled in standby mode in the SAMD core. By default the peripherals and clocks are not enabled in standby mode.

[AT11491: Peripheral Power Consumption in Standby Mode for SAM D Devices](http://www.atmel.com/Images/Atmel-42472-Peripheral-Power-Consumption-in-Standby-Mode-for-SAM-D-Devices_ApplicationNote_AT6490.pdf) is also a good read. I found it while looking into https://github.com/arduino/ArduinoCore-samd/issues/142.

Let me know if this makes sense.

madvoid, 2016-05-31:
Hello @sandeepmistry,

That makes perfect sense, thank you! I'm going to mark this as closed.

closed 2016-05-31 by madvoid

mennamorato, 2018-05-03:
@sandeepmistry If none of the peripherals and clocks are enabled in standby mode, why is that my SAMD21 mini board with all LED's removed still consumes 8mA when using rtc.standbyMode()?  something must be running... other people report less than 100uA with custom code.  Suggestions? I really need to figure this out as I have to get to 200uA max, thanks!

kriti1717, 2019-03-07:
Hi @mennamorato I'm getting the same issue as you. Did you find a solution to it?
I am using delay to wait for my digital pins to get into the write mode before taking a reading. Can I replace my delay function with standby. My guess is standby mode will not work as it shuts down the peripherals but then my standby mode power consumption is 9mA. Please advice

drewfish, 2019-03-07:
The SAMD21 allows a lot of control over which peripherals still run in standby mode. You'll need to read the datasheet for details. It's a little tricky since you need to cross-reference the arduino pin numbers into the EIC signal numbers, then set the correct bits in the EIC configuration register. I also configured the EIC to run off the same clock as the RTC, not sure if that was necessary.

I've used `rtc.standbyMode()` in projects to put the CPU to sleep but still have the EIC (external interrupt controller) peripheral running so that a button press wakes up the CPU. With that I was using 300uA, but some of that might have been other components (voltage regulator, etc) on the devboard I was using.