arduino-libraries/RTCZero #11: rtc.begin() is not needed

lorenzoromagnoli, 2016-07-21:
I was trying to use the rtc library while debugging a different issue related to the wifi library.
my MKR1000 was getting stuck executing the function setEpoch(), because no rtc.begin() was called before. 

The board was getting stuck and I was even unable to upload a new sketch without resetting it. It took me an hour to figure out that I was missing the begin function.

Since you cannot pass any parameter to the begin function, it becomes very easy to forget about it. 

Can we just either move the begin routine to the constructor function or check that the rtc is initialized inside the functions that might get stuck?

sandeepmistry, 2016-07-21:
> Can we just either move the begin routine to the constructor function

To me this is dangerous, because the initialization order is not guaranteed

> check that the rtc is initialized inside the functions that might get stuck?

This is definitely a better option for me.

GabrielNotman, 2016-07-22:
The issue you encountered was due to the methods attempting to write to the _CLOCK_ register before the _RTC_ was configured. Whenever the _CLOCK_ register is modified, the library then waits until the register is synchronised. In cases when the _RTC_ has not been configured, the synchronisation is never completed, resulting in the method entering an endless loop.

In version 1.5.0, this is no longer an issue. If the _RTC_ isn't configured, then any attempts to modify the clock settings, are simply ignored. If you forget to call _rtc.begin()_, the _RTC_ will not function but it will not cause the board to freeze up.

Version 1.5.0 should now be released, however, the version number in _libraries.properties_  has not been updated (it is still 1.4.3).

closed 2017-02-27 by agdl