arduino-libraries/RTCZero #12: BSOD in Arduino create with MKR1000

BallscrewBob, 2016-08-22:
If I upload the sketch from either CREATE or a regular IDE and then open the CREATE serial monitor this will cause a BSOD every time. Amount of time to BSOD can be anywhere from a couple of hours to less than 12.

Using an MKR1000 and AMD quad core, Win 7 X64 Pro.
This is the only sketch that does it but I have to note there are other issues with the MKR randomly halting (2 hrs to a max of 3 days) with some other sketches that often use serial output. However these do not cause BSOD's

Issue posted at the request of sandeepmistry

sandeepmistry, 2016-08-22:
Thanks @BallscrewBob.

@facchinm or @agdl is there a Win 7 machine in the office that someone can use to try to reproduce this issue?

Sketch that causes the issue is: https://create.arduino.cc/editor/Ballscrewbob/7baa3b59-26ae-4bce-8e4e-1ed6f1c10d60/preview which I think is just the [SimpleRTC.ino](https://github.com/arduino-libraries/RTCZero/blob/master/examples/SimpleRTC/SimpleRTC.ino) example.

(from forum post: http://forum.arduino.cc/index.php?topic=408863.msg2887873#msg2887873)

sandeepmistry, 2016-08-31:
Hi @BallscrewBob,

Sorry for the lack of updates on this. Unfortunately, there is no Windows 7 PC in the office for someone to try to reproduce this issue. (PC's were upgraded to Windows 10).

Does this still happen if you use https://github.com/arduino/ArduinoCore-samd/pull/154?

BallscrewBob, 2016-09-14:
Just upped from B88 to B97.

B88 was still BSOD in CREATE mainly under two hours.
Going to try under IDE 1.6.11 to check. Be back in 48 hours (or less if fail)

sandeepmistry, 2016-09-14:
Hi @BallscrewBob,

The behaviour with Create is expected (same as you are experiencing before), since using https://github.com/arduino/ArduinoCore-samd/pull/154 is something only the regular IDE uses. (Just wanted to clarify this, please ignore if you already knew this.)

BallscrewBob, 2016-09-15:
Yup had figured that out. I can do a BSOD on demand almost with CREATE and RTC.

However it also stalls on the MKR in IDE 1.6.11
Start time was 12:11:55
Stall time was 17:41:23

Started a new run and will report when I have details.

sandeepmistry, 2016-09-15:
@BallscrewBob please note https://github.com/arduino/ArduinoCore-samd/pull/154 has been merged in to master, so will be part of the next SAMD core release, which will also get propagated to Create.

You can also switch over the the SAMD core hourly build to preview the next SAMD core release: https://github.com/arduino/ArduinoCore-samd#hourly-builds

Maybe you can run a simpler test like: https://github.com/arduino/ArduinoCore-samd/pull/154#issuecomment-233424068

Then we can know if the board is crashed (LED stops blinking) or it's a problem on the PC side. Also, do you have another PC you can test with?

BallscrewBob, 2016-09-16:
Just over 8 hours before stall in the IDE using the MKR1000 WiFi RTC example.

That blink test passes every time regardless of CREATE or IDE which might indicate an issue with variables serial printing ?
The only sketches that I currently know to fail with some sort of consistency are the RTC and Thingspeak. Thingspeak stops sending data regardless of sensor used so hardware is not the issue.

Ooalis from CREATE is sending me another MKR (f.o.c) so I will be able to do some comparison testing when that arrives.

Swapping to hourly SAMD as suggested.

Don't have my server back up yet and my loaner win 7 box is out at a clients for a while.
Not much else available right now apart from tablets which are all Android base.
Will try get Server 2012 up in next few days as I need it online anyway.

sandeepmistry, 2016-09-16:
> That blink test passes every time regardless of CREATE or IDE which might indicate an issue with variables serial printing ( buffers maybe ? )

Yes, or something's up with the RTC library.

> The only sketches that I currently know to fail with some sort of consistency are the RTC and Thingspeak. Thingspeak stops sending data regardless of sensor used so hardware is not the issue.

I have a pull request for the WiFi101 library that has a bunch of fixes, something else to try regarding Thinkspeak: https://github.com/arduino-libraries/WiFi101/pull/77

Feel free to open an issue in the WiFi101 Github repo regarding this.

> Will try get Server 2012 up in next few days as I need it online anyway.

Thanks, this will help narrow down if it's Windows 7 or issue with your PC. Btw, do you also have a powered USB hub to test with? - I'm wondering if the PC can't provide enough current, just another thought.

BallscrewBob, 2016-09-23:
Hi Sandeep.

Server 2012 up and running with IDE 1.6.12 (all updates are in and using the hourly JSON in additional boards.
MKR attached to USB 2.0 as it will not play at all on USB 3.0 at all. and just spits out an error on trying to upload but Arduinos and USB are known to not play nice together.
Is there a particular sketch you wish to try first ?

sandeepmistry, 2016-09-23:
Wow, that's quite the machine!

Would you be able to try the [SimpleRTC.ino](https://github.com/arduino-libraries/RTCZero/blob/master/examples/SimpleRTC/SimpleRTC.ino) with SAMD core release 1.6.6 (no JSON file). I'm curious to see if the BSOD still occurs.

Thanks for the note on USB 3.0 vs USB 2.0, USB 3.0 seems to be fine on my MacBook Pro .... but that's not really an apples to apples comparison.

BallscrewBob, 2016-09-23:
Confirming upload of the sketch you suggested @ 9.35 EST.
Left serial monitor open on the server but will close it and re=open on occasion.
Also using SAMD 1.6.6 without JSON.

The server is actually quite old and is modified for the SSD's and RAM, its taken a while to get to that spec LOL. Seeing as server 2012 is based around Win 8 / 8.1 it should provide a decent test base for network ability at maybe a deeper level

sandeepmistry, 2016-09-23:
Thanks

> Left serial monitor open on the server but will close it and re=open on occasion.

Please note, with 1.6.6 of the SAMD closing the serial monitor will cause the sketch to block until it is re-open on Windows. The SAMD nightly has a fix for this.

BallscrewBob, 2016-09-23:
K understood so will leave the monitor open and not touch it.
Will just look in on it on occasion to make sure its still running.

BallscrewBob, 2016-09-24:
Had random issues with the IDE. Changed sketch to this one https://create-[dev.arduino.cc/editor/Ballscrewbob/46b43ff7-4e61-440b-99e5-5e6605c04d2e/preview ](url)

Also dropped cores to 1.6.3 as was getting SMD packages unknown errors. but will try tommorow to go back up one at a time. Also had to add the standard JSON back in to get it stable.. There were some lib updates and other core updates which I took so hope it was just a minor glitch.

BallscrewBob, 2016-09-24:
Sketch ran from 5.14 to 6.06 and stalled. Starting to go back up packages.

BallscrewBob, 2016-09-24:
Packages all back up to date and JSON removed restarting sketch.

BallscrewBob, 2016-09-25:
Changed to use the EPOCH example sketch from RTC zero on server

BallscrewBob, 2016-09-27:
No BSOD but stalled just short of 50 hours on the server. Awaiting further destructions

sandeepmistry, 2016-09-27:
Ok, I'd say you try with the nightly SAMD JSON next.

sandeepmistry, 2016-09-28:
@BallscrewBob just a heads up, SAMD core 1.6.7 has been released with the following changes: https://github.com/arduino/ArduinoCore-samd/blob/master/CHANGELOG

It includes the USB fixes that you've tested with previously.

BallscrewBob, 2016-10-04:
OK have not had time to check it out BUT I think we may be able to close the RTC issue BSOD as I have just hit the same BSOD issue without an MKR and without any RTC items.

Pretty sure it is down to an issue with USBSER.SYS which may be outside the bounds of this issue and may need a different approach or git issue. Boards active at the time were a 101, Mega, and UNO all using different serial statements and sketches.

sandeepmistry, 2016-10-04:
Ok, I'll close it for now then. `USBSER.SYS` makes it feel like a general Windows OS issue if it impacts all boards. Have you done any testing with different versions of windows or other PCs?

closed 2016-10-04 by sandeepmistry

BallscrewBob, 2016-10-06:
Yes it locked up on server 2012 requiring a hard restart a few times.

As server 2012 is win 8 / 8.1 based but hardened for server duties and
network and I think that's the only reason it didn't BSOD.

On the win 7 box usbser.sys is ONLY used by Arduino and nothing else.
It is ALWAYS listed as the cause during the BSOD.
Microsoft say its a driver issue which puts me right back to Arduino.

If I google for "bsod usbser.sys" I end up with quite a few Arduino related
items so it seems it is an issue, but I feel that it is one that nobody may
want to touch as it is not a common issue.
Fact that I can replicate it on two different hardware bases (INTEl and
AMD) indicates it is not hardware specific in terms of computer hardware.
can also replicate it across a variety of board now too which means we were
chasing a ghost on RTC ZERO.

sandeepmistry, 2016-10-06:
> Microsoft say its a driver issue which puts me right back to Arduino.

I'm not a Windows expert by any means, however this page suggests Microsoft is in charge of the driver:

https://msdn.microsoft.com/en-ca/library/windows/hardware/dn707976(v=vs.85).aspx

>> USB serial driver (Usbser.sys)
>> Microsoft-provided in-box driver (Usbser.sys) for your Communications and CDC Control device.

BallscrewBob, 2016-10-07:
> dn707976(v=vs.85).aspx

On most cases you will be correct however in this case usbser is not a
Microsoft one.

sandeepmistry, 2016-10-07:
Who provides it then? Is there anyway you can go back to the Microsoft version?

I didn't find any copies of `usbser.sys` in the [driver's folder on Github](https://github.com/arduino/Arduino/tree/master/build/windows/dist/drivers). [arduino.inf](https://github.com/arduino/Arduino/blob/master/build/windows/dist/drivers/arduino.inf) just references `usbser.sys`.