arduino-libraries/RTCZero #15: Compile errors in either DEV create or regular CREATE

BallscrewBob, 2016-09-26:
Compile errors in either DEV CREATE or regular CREATE

Using board 'mega' from platform in folder: /opt/cores/arduino/avr
Using core 'arduino' from platform in folder: /opt/cores/arduino/avr
Detecting libraries used...
"/opt/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/opt/cores/arduino/avr/cores/arduino" "-I/opt/cores/arduino/avr/variants/mega" "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/sketch/SimpleRTC_copy.ino.cpp" -o "/dev/null"
"/opt/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/opt/cores/arduino/avr/cores/arduino" "-I/opt/cores/arduino/avr/variants/mega" "-I/tmp/webide_tmp186101/libraries/RTCZero/src" "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/sketch/SimpleRTC_copy.ino.cpp" -o "/dev/null"
"/opt/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/opt/cores/arduino/avr/cores/arduino" "-I/opt/cores/arduino/avr/variants/mega" "-I/tmp/webide_tmp186101/libraries/RTCZero/src" "/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.cpp" -o "/dev/null"
"/opt/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/opt/cores/arduino/avr/cores/arduino" "-I/opt/cores/arduino/avr/variants/mega" "-I/tmp/webide_tmp186101/libraries/RTCZero/src" "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/sketch/SimpleRTC_copy.ino.cpp" -o "/dev/null"
WARNING: library RTCZero claims to run on [samd] architecture(s) and may be incompatible with your current board which runs on [avr] architecture(s).
Generating function prototypes...
"/opt/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/opt/cores/arduino/avr/cores/arduino" "-I/opt/cores/arduino/avr/variants/mega" "-I/tmp/webide_tmp186101/libraries/RTCZero/src" "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/sketch/SimpleRTC_copy.ino.cpp" -o "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/preproc/ctags_target_for_gcc_minus_e.cpp"
"/opt/arduino-builder/tools/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/opt/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-I/opt/cores/arduino/avr/cores/arduino" "-I/opt/cores/arduino/avr/variants/mega" "-I/tmp/webide_tmp186101/libraries/RTCZero/src" "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/sketch/SimpleRTC_copy.ino.cpp" -o "/tmp/tmp/webide_tmp186101/SimpleRTC_copy/sketch/SimpleRTC_copy.ino.cpp.o"
In file included from /tmp/webide_tmp186101/SimpleRTC_copy/SimpleRTC_copy.ino:17:0:
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:32:26: error: 'RTC_MODE2_MASK_SEL_OFF_Val' was not declared in this scope
MATCH_OFF = RTC_MODE2_MASK_SEL_OFF_Val, // Never
^
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:33:26: error: 'RTC_MODE2_MASK_SEL_SS_Val' was not declared in this scope
MATCH_SS = RTC_MODE2_MASK_SEL_SS_Val, // Every Minute
^
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:34:26: error: 'RTC_MODE2_MASK_SEL_MMSS_Val' was not declared in this scope
MATCH_MMSS = RTC_MODE2_MASK_SEL_MMSS_Val, // Every Hour
^
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:35:26: error: 'RTC_MODE2_MASK_SEL_HHMMSS_Val' was not declared in this scope
MATCH_HHMMSS = RTC_MODE2_MASK_SEL_HHMMSS_Val, // Every Day
^
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:36:26: error: 'RTC_MODE2_MASK_SEL_DDHHMMSS_Val' was not declared in this scope
MATCH_DHHMMSS = RTC_MODE2_MASK_SEL_DDHHMMSS_Val, // Every Month
^
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:37:26: error: 'RTC_MODE2_MASK_SEL_MMDDHHMMSS_Val' was not declared in this scope
MATCH_MMDDHHMMSS = RTC_MODE2_MASK_SEL_MMDDHHMMSS_Val, // Every Year
^
/tmp/webide_tmp186101/libraries/RTCZero/src/RTCZero.h:38:26: error: 'RTC_MODE2_MASK_SEL_YYMMDDHHMMSS_Val' was not declared in this scope
MATCH_YYMMDDHHMMSS = RTC_MODE2_MASK_SEL_YYMMDDHHMMSS_Val // Once, on a specific date and a specific time
^
Using library RTCZero at version 1.4.3 in folder: /tmp/webide_tmp186101/libraries/RTCZero

BallscrewBob, 2016-09-26:
Compiles fine for MKR but when compiling for 101 or MEGA the errors come up.

sandeepmistry, 2016-09-26:
The RTCZero library is only mean't for SAMD boards (Zero, MKR1000).

cc/ @smellai @matteosuppo

sandeepmistry, 2016-09-27:
I'm going to close this for now, it's the same behaviour as well you try to use the library on a non-SAMD board in the Java IDE. Note, there is a warning in the output:

```
WARNING: library RTCZero claims to run on [samd] architecture(s) and may be incompatible with your current board which runs on [avr] architecture(s).
```

closed 2016-09-27 by sandeepmistry

epugar, 2018-12-03:
I have this error too: _

epugar, 2018-12-03:
I have this error too:
"error: 'RTC_MODE2_MASK_SEL_OFF_Val' was not declared in this scope

     MATCH_OFF          = RTC_MODE2_MASK_SEL_OFF_Val,          // Never"
same goes for every minute, etc.
My question: I downloaded code from Arduino Project Hub and I'm just testing it. If I connect my Arduino MKR1000 will this still be an error? I'm just new to this area so sorry if it a dumb question.

per1234, 2018-12-03:
@epugar which board do you have selected? As was already stated, this library is only compatible with the Arduino SAMD Boards. If you try to compile for a board of any other architecture, you'll get that error.

The MKR1000 is SAMD architecture so if you select that board you should not get this error. For compilation, it doesn't matter whether you have the MKR1000 connected or not. That only matters for uploading.

epugar, 2018-12-03:
@per1234  That board was not selected. As I said, I'm completely new to this. I would never think about checking that. Thank you!