arduino-libraries/RTCZero #16: High latency when reading the date-time

sergiuburian, 2016-10-11:
I am using an M0 board from Adafruit,with the RTCZero library. I innitialize and set the RTCZero from NTP servers at start-up and then read it in the loop in order to timestamp (as string) my sensor data log entries (date / time from RTCZero and millis() to get millisecond kind-of accuracy ). The data acquisition frequency is 100 Hz. 
The problem is, that in order to read the time for this very simple and common purpose, I need to call getYear(), getMonth(), ... , getSeconds() one after each other; I avoid getEpoch, as the mktime is expensive and I need to parse back the epoch in human readable format. I only have 10 millis per loop . Each of those subsequent calls will issue a read request and wait for RTC->MODE2.STATUS.bit.SYNCBUSY to complete; this leads to a total time of about 40 millis, which is  lot (even an SD card is able to write a text line every 2 millis). 

My suggestion is to add a method tm RTCZero getDateTime(), similar with uint32_t RTCZero::getEpoch()  but without calling the mktime. This issues one single read request and than extracts all six values from the registry, so it should take less time than the 6 read requests issued when using the discrete methods. Or, if you want to really be nice, you can also add a method that formats a date/time string according to a passed format :)

Thanks in advance,

Sergiu

GabrielNotman, 2016-11-21:
If you want to read the values directly they can be accessed via the _RTC->MODE2.CLOCK.reg_ register. You can see how this is done in the _getEpoch()_ method.

If you skip the synchronisation request, it will read quicker. However, if you attempt to read it during a background register update it will cause the bus to stall for about 5ms. This can cause a lot of issues as it can block interrupts. If you have data streaming into a UART, this can result in dropped characters as the interrupt which copies the received data from the SERCOM register into the ring buffer is delayed to the point where another character has already arrived.

sergiuburian, 2016-11-22:
Hi Gabriel,

Thanks for the hint. Of course I can read myself the registers, as in getEpoch. I just thought, this would be a very useful function for all those who use this library in combination with data loggers.
Thanks,
Sergiu