arduino-libraries/RTCZero #21: rtc.standbymode() only working once

Sune89, 2016-11-23:
I am doing project following the example from http://forcetronic.blogspot.co.uk/2016/09/reducing-power-consumption-on-arduino_24.html.

My Genuino MKR1000 goes to standbymode, and wakes up when motion is detected on the PIR sensor. However, for some reason I cannot make it go back into standby mode. I have tried several different ways of debugging this, adding the standbymode at different places in the code, without any luck. Could the issue be in the library? 

<pre><code>
#include &ltRTCZero.h>

/* Create an rtc object */
RTCZero rtc;

/* Change these values to set the current initial time */
const byte seconds = 0;
const byte minutes = 00;
const byte hours = 00;

/* Change these values to set the current initial date */
const byte day = 23;
const byte month = 11;
const byte year = 16;

void setup() {
 Serial.begin(9600);
 
 delay(5000);

 pinMode(1, INPUT);
 //LED connected to digital pin x
 pinMode(2, OUTPUT);
 //PIR sensor connected to digital pin x
 pinMode(LED_BUILTIN, OUTPUT); //set LED pin to output
 digitalWrite(LED_BUILTIN, LOW); //turn LED off

 rtc.begin(); //Start RTC library, this is where the clock source is initialized

  rtc.setTime(hours, minutes, seconds); //set time
  rtc.setDate(day, month, year); //set date
  pinMode(1, INPUT);   //already declared before but maybe needed again?
  attachInterrupt(1, ISR, HIGH); //creates an interrupt source on PIR activation
  
  rtc.standbyMode();
}
void ISR()
{
  Serial.println("Here");
  digitalWrite(2, HIGH);
  delay(250);
  digitalWrite(2, LOW);
  delay(250);
  digitalWrite(2, HIGH);
  delay(250);
  digitalWrite(2, LOW);
  delay(250);
  digitalWrite(2, HIGH);
  delay(250);
  digitalWrite(2, LOW);
  rtc.standbyMode();
}</code></pre>

GabrielNotman, 2016-12-13:
Using delay in an ISR will most likely cause issues, as will printing to certain types of streams.

In the above example, I would recommend moving all of the ISR code to _loop()_. When the interrupt is triggered, it will run the empty ISR and then continue from the last line of _loop()_. Then it will run _loop()_ again until it hits the call to standby where the board will enter sleep mode.

EricaSwift, 2018-01-09:
@GabrielNotman This was great advice. I tried leaving the ISR loop completely blank and moved my code into the main loop. Worked immediately.

agdl, 2018-10-24:
Can we close this issue?

sslupsky, 2018-10-24:
FWIW, it appears to me the call to rtc.standbyMode() in the ISR was the original problem.  Recursive standby ... seems like a definite no-no.

closed 2019-04-16 by agdl