arduino-libraries/RTCZero #25: Adafruit Feather M0 won't wake up?

softweyr, 2017-01-27:
I have an Adafruit Feather M0 RFM69HCW that has become redundant from another project, so I thought I'd make an outdoor temperature sensor from it.  

The idea is to wake up once a minute, sample the temperature from a DHT-11 I have lying around, blast it out over the RFM radio, then go to sleep.  It's all working moderately well, except I can't get the M0 to 'wake up' once I put it in standby.  The ISR is called, but `loop()` never runs again.  

I noticed something in the code about standby being problematic when the onboard USB is connected, and this still has a lot of serial debug code in it.  Should I give it a try unplugged, perhaps redirecting the prints to Serial1 with my trust FTDI board?

frankleonrose, 2017-09-11:
Take a look [here](https://github.com/cavemoa/Feather-M0-Adalogger/blob/master/SimpleSleepUSB/SimpleSleepUSB.ino#L36): 
```
Serial.end();
USBDevice.detach(); // Safely detach the USB prior to sleeping
rtc.standbyMode();    // Sleep until next alarm match
USBDevice.attach();   // Re-attach the USB, audible sound on windows machines
```
@cavemoa makes a comment somewhere that this was critical to getting sleep working while USB connected. Also that, depending on your OS and tools, you may have to restart your serial monitor after the `detach`.

softweyr, 2018-11-09:
I finally got back to this.  I'll take a look at the above, or just blink and LED or something to indicate progress.  Thanks!

closed 2018-11-09 by softweyr