arduino-libraries/RTCZero #28: Using RTCZero and TimeLib.h doesn't compile

dIoTte, 2017-02-27:
Any idea why it is not possible to use both libraries at the same time?

#include <TimeLib.h>
#include <RTCZero.h>
void setup() {
}
void loop() {
}

sandeepmistry, 2017-03-01:
Hi @dIoTte,

This is because `TimeLib.h` contains a file named [Time.h](https://github.com/PaulStoffregen/Time/blob/master/Time.h) which on case insensitive filesystems (used by default on Mac and Windows) conflicts with the POSIX `time.h` header the `RTCZero` library uses.

We could ask the `TimeLib` maintainers to rename the `Time.h` file via a pull request. @facchinm @cmaglie what do you think?

facchinm, 2017-03-02:
@sandeepmistry totally agree, but we should think about some sort of wrapper to avoid troubles for someone who includes `Time.h`

dIoTte, 2017-05-08:
Hello everyone, is it still in the plan to fix this problem?

Thanks,
Seb

sandeepmistry, 2017-05-10:
@dIoTte I've opened to following issue to continue the discussion https://github.com/PaulStoffregen/Time/issues/74

sandeepmistry, 2017-05-11:
Hi @dIoTte,

Based on the discussion in https://github.com/PaulStoffregen/Time/issues/74#issuecomment-300629882, @PaulStoffregen is not keen on removing `Time.h` from the `TimeLib` library for compatibility with older IDE versions.

As a work around, we would suggest you manually remove `Time.h` from `<SKETCHBOOK>/libraries/Time/` if you wish to use both `TimeLib` and `RTCZero` in your sketch.

cc/ @cmaglie

DiegoD4i, 2019-03-16:
Sorry, I´ve a problema with this library. 
When I use  GMT-3 setup, the library Works well… but near mitnight, returns wrong day and time.
Could you help me please.
Thanks a lot

thewoz, 2020-06-22:
I have the same problem.
So the solution is to manually remove Time.h from <SKETCHBOOK>/libraries/Time/?