arduino-libraries/RTCZero #33: Freeze after Standby

jamesfahlbusch, 2017-12-05:
After waking from the RTCZero Standby the system freezes when I attempt to use interrupts. Is something not being reset after standby?
I am using a library called Retick that throttles the loop and utilizes TCC1 and it is getting stuck at the first instance of _WFI() after standby. Are the interrupts not being cleared?

#include <RTCZero.h>
#include "Arduino.h"
#include "retick.h"

#define RED 13 // Red LED on Pin #13
#define GREEN 8 // Green LED on Pin #8
#define VBATPIN A7    // Battery Voltage on Pin A7

// re-defines USB serial from M0 chip so it appears as regular serial
#if defined(ARDUINO_SAMD_ZERO) && defined(SERIAL_PORT_USBVIRTUAL)
  // Required for Serial on Zero based boards 
  #define Serial SERIAL_PORT_USBVIRTUAL
#endif

/////////////// Key Settings ///////////////////

byte hours = 0;
byte minutes = 0;
byte seconds = 0;
byte day = 1;
byte month = 1;
byte year = 17; 
byte Start = 0;
byte retickRate = 2000;

/////////////// Global Objects ////////////////////
RTCZero rtc;          // Create RTC object

//////////////    Setup   ///////////////////
void setup() {
  rtc.begin(); // initialize RTC
  Serial.begin(9600);
  while (!Serial); 
  Serial.println("Setup");
  rtc.begin(); // initialize RTC
  rtc.setTime(hours, minutes, seconds);
  rtc.setDate(day, month, year);
  rtc.setAlarmTime(hours, minutes, seconds + 20);
  rtc.setAlarmDate(day, month, year); 
  rtc.enableAlarm(rtc.MATCH_YYMMDDHHMMSS); 
 
  Serial.println("Entering Standby");
  
  rtc.standbyMode();
 
  rtc.disableAlarm(); 
  rtc.detachInterrupt();
  rtc.updateAlarms();
  delay(5000);
 

}  

/////////////////////   Loop    //////////////////////
void loop() {
  //Serial.begin(9600);
  Serial.println("Entering Loop");
  /*  Each run-through of tick will be 20ms (50Hz)
   *  If it takes less than 20ms, the CPU will sleep
   *  If it takes more, it immediately goes into next 
   *  tick once the previous finishes
   */
  retick(retickRate);
}

/////////////////////   ReTick    //////////////////////
void tick() {
 
    Serial.println("Entering retick loop");
    while(1) {
    uint8_t i;
    for (i=0; i<2; i++) {
      digitalWrite(13, HIGH);
      delay(500);
      digitalWrite(13, LOW);
      delay(500);
    }
    for (i=2; i<10; i++) {
      delay(300);
    }
  }

}

marcobrianza, 2019-01-16:
I had similar periodic random issues and proposed this fix
https://github.com/arduino-libraries/RTCZero/pull/42

can you please try it?

wjpavalko, 2019-01-16:
Ok.  I'll see if I can.

I was able to prevent the problem by inserting a short delay (10 ms) into
the code right before the call for sleep.  Not sure why it works, but it
solved the problem I was having.  All buoys with that code have now worked
continuously without the hang I was seeing.

--Wayne

closed 2019-04-16 by agdl