arduino-libraries/RTCZero #39: Carry over not increment of current time date in example

BallscrewBob, 2018-08-04:
Using the example [https://create.arduino.cc/editor/ballscrewbob/4aa164b5-3012-4361-a7ff-d6a48eadea90/preview](url)

The time carries over but does not INCREMENT the date and reset the clock to zero.

Also seen in the forum [https://forum.arduino.cc/index.php?topic=557099.new#new](url)

Wijnbouwer, 2020-05-19:
Had the same problem. I'm working on a MKR1400 which tends to hang after approx 24 hr (known issue with gsmclient), By monitoring the system overnight, I noticed the same problem. It corrected itself a 25:59 hr -> 02:00.
If not solved: is there a work-around? Since I drop the collected data in mysql, the data with weird hours gets rejected and is lost. I can not afford to loose two hours of data every day.
thx ahead !
Kris

luisdamed, 2020-10-17:
Hello @Wijnbouwer and @BallscrewBob 

I found the same issue and it seems that this hasn't been solved yet (please correct me if I am wrong!). I noticed that also the month and day, and year are not being updated after midnight until 25:59 hr -> 02:00.

I have this workaround solution that worked OK for me, but I am sure that the best way would be to solve the issue at a library level. I am quite new to Arduino (and programming, in general) so this probably not a really "clean" fix.

`#include <RTCZero.h>
RTCZero rtc;

/* Change these values to set the current initial time */
const byte seconds = 50;
const byte minutes = 59;
const byte hours = 23;

/* Change these values to set the current initial date */
const byte day = 31;
const byte month = 12;
const byte year = 20;

void setup() {
  Serial.begin(9600); // establish communication parameters with serial port
  rtc.begin();

  // Set the time
  rtc.setHours(hours);
  rtc.setMinutes(minutes);
  rtc.setSeconds(seconds);

  // Set the date
  rtc.setDay(day);
  rtc.setMonth(month);
  rtc.setYear(year);
}

void loop() {
  // put your main code here, to run repeatedly:
  int hour = (rtc.getHours() );
  int mins = (rtc.getMinutes());
  int sec = (rtc.getSeconds());
  int day = (rtc.getDay());
  int month = (rtc.getMonth());
  int year = (rtc.getYear());
  int daysInFebruary = 28 + leapyear(year);
  if (hour >= 24) {

    //Workaround for RTC Zero library ver 1.6.0 bug #39
    hour = (hour - 24);
    day = (day + 1);

    //Move from one month to another or change year

      if (month == 1||month == 3||month == 5||month == 7||month == 8||month == 10){
        //If the month has 31 days, change month when the day is already 31 (except for new year)
        if (day > 31) {
          day = 1;
          month = month + 1;
        }
      }
      else if  (month ==4||month ==6||month ==9||month ==11){
        //If the month has 30 days, change month when the day is already 30
        if (day > 30) {
          day = 1;
          month = month + 1;
        }
      }
      else if (month==2){
        //Move from February to March
        if (day > daysInFebruary) {
          day = 1;
          month = month + 1;
        }
      }
      else if (month==12){
        //Move from December to January of the next year
        if (day > 31) {
          day = 1;
          month = 1;
          year = year + 1;
        }
      }
    

  }
  String timestamp = (use2digits(day) + "/" + use2digits(month) + "/" + String(year) + " " + use2digits(hour) + ":" + use2digits(mins) + ":" + use2digits(sec));

  Serial.print(timestamp + "\n");
  delay(1000);
}

String use2digits(int number) {
  if (number < 10) {
    return "0" + String(number); // print a 0 before if the number is < than 10
  }
  return String(number);`