arduino-libraries/RTCZero #40: Feature:  Periodic alarms between 1 sec and 1 minute

sslupsky, 2018-09-23:
The smallest period with the current version is 1 minute.  It would be useful to be able to set periodic alarms that fire more frequently.  

Perhaps something like MATCH_MS (match milliseconds) ?  That would provide a 1 second period.  I do not require anything fast than that.  

It would be useful to be able to set the period to an arbitrary period between 1 and 59 seconds.

sslupsky, 2019-05-24:
I would like to submit a PR for this issue so I would like to get some feedback.  I see two ways to accomplish a better than 1 minute periodic interrupt.

1.  Set a new ALARM timeout in the RTC hander.
2.  Use the Periodic Events

The first method is the technique I presently use.  It has the disadvantage that it requires a write to the alarm register.  This write causes a SYNCBUSY and requires about 6ms to complete.  This is a concern when trying to save power.

The second method involves the Periodic Events generated from the RTC Prescaler.  I think the SYNCBUSY delay can be avoided.

Are there any other ways to accomplish this?  Any other feedback?

ravelab, 2019-08-14:
If SYNCBUSY delay can be avoided, that would be great.