arduino-libraries/RTCZero #50: Crystal-less implementation time issue

ubidefeo, 2019-06-20:
I have used this code to successfully set the device to sleep and wake up after 20 seconds, but on the Nano 33 IoT it won't work and I guess it's due to how seconds are turned into ticks and the fact that it doesn't tick at 32KHz. Just guessing :)
Works on MKR Zero, MKR WiFi 1010 and MKR GSM 1400

```c
RTCZero realTimeClock;

void enterSleep(){
  realTimeClock.begin();
  uint32_t rtcEpoch = realTimeClock.getEpoch();
  realTimeClock.setAlarmEpoch(rtcEpoch + 20);
  realTimeClock.enableAlarm(realTimeClock.MATCH_HHMMSS);

  realTimeClock.attachInterrupt(ISR);
  realTimeClock.standbyMode();
}

void ISR(){
  playWakeUpSequence = true;
  realTimeClock.disableAlarm();
  realTimeClock.setTime(hours, minutes, seconds);
}
```

pnndra, 2019-06-20:
Hi. Since we don't have a 32khz crystal rtc can't be used on this board

ubidefeo, 2019-06-20:
Ran a test not dependant on Serial output by blinking the onboard LED 3 times at wakeup and it indeed worked.
So it might just be a case of bus stall as referenced in #44 .
I'll close this one but maybe someone can take a look if it's the case to merge that PR :)

closed 2019-06-20 by ubidefeo