arduino-libraries/RTCZero #52: sleep one extra second?

ravelab, 2019-08-14:
I set it up like this:

  rtc.begin();
  rtc.enableAlarm(rtc.MATCH_SS);

loop() {
  LED OFF
  rtc.setSeconds(0);
  rtc.setAlarmSeconds(0);
  rtc.standbyMode();
  LED ON
}

I noticed that if I setAlarmSeconds(0), it would sleep 1 second, if I setAlarmSeconds(4),it would sleep 5 seconds. So the total sleeptime is alarmSeconds - seconds + 1. Is that the expected behavior?

Thanks

moeskerv, 2020-02-04:
I can confirm this. Basically sleeping for some seconds only seems to always add 1s extra:

rtc.setSeconds(0);
rtc.setAlarmSeconds(0);
rtc.enableAlarm(rtc.MATCH_SS); 
rtc.standbyMode(); 

Above results in a sleep for 1s.

rtc.setSeconds(0);
rtc.setAlarmSeconds(1);
rtc.enableAlarm(rtc.MATCH_SS); // Match seconds
rtc.standbyMode();    // Sleep until next alarm match

Above results in a sleep for 2 seconds. 
And so on.

I'm using version RTCZero 1.6.0 on a Feather M0.

moeskerv, 2020-02-04:
Interestingly the ArduinoLowPower (which uses the RTC too) lib does the same on my Feather M0

LowPower.sleep(0); -> sleeps for 1s
LowPower.sleep(1000); -> sleeps for 2s

I tested with the TimedWakeup example: https://www.arduino.cc/en/Tutorial/LowPowerTimedWakeup

moeskerv, 2020-02-04:
Might be a special behavior of the Adafruit M0 based Feathers?