arduino-libraries/RTCZero #57: Feature Request: Add Support for SAMD51 MCUs

TheShadowhawke, 2020-02-05:
Could support be added to this library for the SAMD51 series of MCUs? Thanks in advance.

DCelectronics, 2020-11-27:
@TheShadowhawke and @per1234 
I tried this on the Feather M4, Metro M4, and Grand Central M4, and it worked on all three different boards. 
https://github.com/ericbaril72/RTCZero/tree/SAMD51_E54
I installed the latest from Arduino library manager (current version 1.6.0) and overwrote only the [RTCZero.cpp](https://github.com/ericbaril72/RTCZero/blob/SAMD51_E54/src/RTCZero.cpp) file.
This modified library did not work on ItsyBitsy M4 because it is crystal-less. It may be possible to add support for crystal-less SAMD51.

drewfish, 2020-11-29:
I have a version which works on the Adafruit ItsyBitsy M4, I could make a pull request if repo owner(s) think it might be valuable. (I'd first want to test it out with SAMD21 crystal & crystalless boards also, of course.)