arduino-libraries/RTCZero #62: Device eventually stops waking from sleep using RTCZero 1.6.0

pomplesiegel, 2020-07-13:
Hello Everyone,

Great library, and it has served me well over the past couple years. Thank you for your hard work. 

I recently encountered an issue with my Adafruit Feather M0-based board (SAMD21) when combining LMIC with RTCZero (same combo I have used successfully for two years). Essentially, RTCZero is used to sleep in-between Lorawan transmits. 

It seems there is an issue with the combination of LMIC v3.2 (or later) and RTCZero 1.6.0. It's a bit sneaky, as if you test each library's functionality individually, everything functions correctly. However, when combined, the device goes to bed after about a day of sleeps (at transmits every 15s, for testing reproducing purposes). 

See cross-posted issue at LMIC: https://github.com/mcci-catena/arduino-lmic/issues/588

tl;dr: It seems that downgrading RTCZero from 1.6.0 -> 1.5.2 solved the problem without breaking anything. Does this ring a bell for anyone? 

Please note that in the LMIC issue, I note that the millis() values eventually turn negative. That is no longer happening with 1.5.2. 

Thanks!

moeskerv, 2020-10-09:
Seems that I see a similar behavior using RTCZero in combination with the Radiohead library. I also make use of a  Adafruit Feather M0-based with Lora 433 module on it. Also BME280 library is in use. While it was working for some years now, it started to stop from waking up after some random time (hours) with current build. I will do some tests now to see if this can be fixed using RTCZero 1.5.2.

pomplesiegel, 2020-10-09:
@moeskerv, Great to hear someone else is experiencing a similar behavior. Yes, please let us know the result of your test.

moeskerv, 2020-10-13:
Some results from my tests:

- Using RTCZero V1.6.0 in my setup causes the device to not wake up any more after a few hours of operation, reproducible on two devices
- Using RTCZero V1.5.2 in the same setup now runs for several days already on both devices

I now started to add the changes done between the two versions step by step to identify what change caused the issue. This will take some days....

pomplesiegel, 2020-10-13:
@moeskerv, got it. This is my exact same experience as well. I'm glad to hear it is consistent between our setups and applications. 

Great, I look forward to your findings! Thank you

moeskerv, 2020-10-15:
As it seems this part of the changes done between version 1.5.2 and 1.6.0 causes the issue:

+// Default date & time after reset
+#define DEFAULT_YEAR    2000    // 2000..2063
+#define DEFAULT_MONTH   1       // 1..12
+#define DEFAULT_DAY     1       // 1..31
+#define DEFAULT_HOUR    0       // 1..23
+#define DEFAULT_MINUTE  0       // 0..59
+#define DEFAULT_SECOND  0       // 0..59
+
 voidFuncPtr RTC_callBack = NULL;
 
 RTCZero::RTCZero()
@@ -83,12 +91,17 @@ void RTCZero::begin(bool resetTime)
   RTCenable();
   RTCresetRemove();
 
-  // If desired and valid, restore the time value
-  if ((!resetTime) && (validTime)) {
+  // If desired and valid, restore the time value, else use first valid time value
+  if ((!resetTime) && (validTime) && (oldTime.reg != 0L)) {
     RTC->MODE2.CLOCK.reg = oldTime.reg;
-    while (RTCisSyncing())
-      ;
   }
+  else {
+    RTC->MODE2.CLOCK.reg = RTC_MODE2_CLOCK_YEAR(DEFAULT_YEAR - 2000) | RTC_MODE2_CLOCK_MONTH(DEFAULT_MONTH) 
+        | RTC_MODE2_CLOCK_DAY(DEFAULT_DAY) | RTC_MODE2_CLOCK_HOUR(DEFAULT_HOUR) 
+        | RTC_MODE2_CLOCK_MINUTE(DEFAULT_MINUTE) | RTC_MODE2_CLOCK_SECOND(DEFAULT_SECOND);
+  }
+  while (RTCisSyncing())
+    ;
 
   _configured = true;
 }

I will dig deeper to check which part exactly it is.

ryandamico, 2021-09-20:
Great sleuthing. Was there any progress here?

pomplesiegel, 2021-09-20:
@ryandamico, I'm still using 1.5.2 as a result in order to avoid the issue.

Kealper, 2022-01-26:
Tossing my hat into the ring, I've got almost the same setup and noticed the same failure to wake from sleep after a few hours with 1.6.0, going to try 1.5.2 now to see if it stays working correctly. Hardware is the Adafruit Feather M0 with the built-in SX1276 LoRa module using RadioHead for interfacing with the module.

botkero, 2022-07-07:
I use RTCZero and MKRWAN and both separately work fine.
But together they do not work. RTCZero actually works normally? but it then stops at MKRWAN.
Unfortunately, the 1.5.2 version did not help me. Is there any way to verify that I am really using 1.5.2? According to the IDE 1.5.2 is installed.