arduino-libraries/RTCZero #89: Feature request: Adding a function using the register "FREQCORR" for clock correction

Abravius, 2022-06-28:
Hi,

The RTC in my chip isn't really accurate. I think I'm not the only one

After some research in the CCP file, I haven't find any function that uses the "FREQCORR" register.

I think it would be very important to add a function to correct the clock.

Datasheet :

https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/Atmel-42181-SAM-D21_Datasheet.pdf

Register in the page 262.

Thank you in adavance !

closed 2022-06-29 by Abravius

Abravius, 2022-06-29:
I've this solution :

`//Function used to correct the RTC clock

//<_Sign> :   If the correction must be positive (false) or negative (true)
//<_Value> :  Value to put in the register from 1 to 127.
void SetClockCorrection(bool _Sign, byte _Value)
{
  RTC->MODE2.FREQCORR.reg = (byte)_Sign << 8 || _Value;
}`

Abravius, 2022-06-30:
After 24 hours, same problem. I don't know why...

pe-jot, 2023-10-30:
> After 24 hours, same problem. I don't know why...

Your code shifts `_Sign` one bit too far. Anyway, I'm wondering why you don't simply use a `signed char` or `int8_t` type.
Furthermore, as far as I read the datasheet, there's need for write synchronization, thus you'll have to place something like
`while(RTC->MODE2.STATUS.bit.SYNCBUSY);` before writing the register.